a1exdandy/checkm8-a5 #1: unable to compile in windows

amats211, 2020-07-03:
good day sir,

trying to compile in windows but it says,

class USB' has no member named 'ctrlReq_SETUP'; did you mean 'ctrlReq'?

[image]

any thoughts?

farzad9001, 2020-07-03:
You don't apply patch correctly

ramsdv31, 2020-08-06:
Try to use Linux terminal for windows

Eduard0098, 2020-08-16:
hi one a cuestion, how to patch in windows.?

Arbus228, 2020-10-04:
> Вы неправильно применили патч

Hello, I'm new to this, can you please tell me how to apply the patch correctly?

Arbus228, 2020-10-04:
> привет одна подсказка, как в windows патчить.?

У меня такой же вопрос, если вы поняли как это делать, то можете пожалуйста мне рассказать? :(