a1exdandy/checkm8-a5 #2: "usb init error" when running

Guinea1, 2020-07-03:
Hello, I have followed all the instructions and have the correct hardware (arduino uno and MAX3421E based USB host shield), I also correctly patched the usb host library files, but when I try to use this with my iPhone 4s (8940) in DFU mode and watch the serial monitor on my PC, it says "checkm8 started" and then instantly "usb init error", I have also tried this without it connected to my PC, but have no luck as the LED never lights up at the end. What could the issue be?

real-acmkan, 2020-07-23:
any ideas??

ramsdv31, 2020-08-06:
Clone arduino. It happened to one of my unos

real-acmkan, 2020-08-06:
> Clone arduino. It happened to one of my unos 

Actually I discovered it means the USB Host Shield isn't receiving power. It can be fixed by soldering the two pads near the top of the board. Follow this article: [Link](https://esp8266-notes.blogspot.com/2017/08/defective-arduino-usb-host-shield-boards.html?m=1)

Guinea1, 2020-08-11:
Yep, soldering those 3 pads worked great!

real-acmkan, 2020-08-11:
> Yep, soldering those 3 pads worked great! (Though I still have the issue where it just spits out zeros and says "Done!")

Glad to hear it! Send a screenshot of the error and I can help you

Guinea1, 2020-08-11:
> Glad to hear it! Send a screenshot of the error and I can help you

Its exactly this, hopefully there is a solution: https://github.com/a1exdandy/checkm8-a5/issues/4

real-acmkan, 2020-08-11:
> > Glad to hear it! Send a screenshot of the error and I can help you
> 
> 
> 
> Its exactly this, hopefully there is a solution: https://github.com/a1exdandy/checkm8-a5/issues/4

Re-upload the code to the arduino, make sure the USB Host Shield is connected when you upload it. Reboot the device and then boot the device normally and let it sit for 3 minutes. Connect the device to your Mac and have iTunes open. Put it into DFU and then disconnect it from the Mac and quit iTunes. Then connect it to the arduino and then plug the arduino into the Mac. Once it's done you should be good and can now disconnect everything.

If this does not work, go into recovery mode and reinstall iOS. Then proceed to follow the above steps after the device has turned on. No need to restart the idevice.

ramsdv31, 2020-08-12:
> Re-upload the code to the arduino, make sure the USB Host Shield is connected when you upload it. Reboot the device and then boot the device normally and let it sit for 3 minutes. Connect the device [...]

Please post your arduino monitor. Are you getting a bunch oz ZEROS and a DONE at the end?

ramsdv31, 2020-08-12:
> Actually I discovered it means the USB Host Shield isn't receiving power. It can be fixed by soldering the two pads near the top of the board. Follow this article: [...]

Naaa that didn't work either

real-acmkan, 2020-08-12:
> Please post your arduino monitor. Are you getting a bunch oz ZEROS and a DONE at the end?

I can't as I solved the issue for myself. Look at the referenced issue for an example of what I had.

Guinea1, 2020-10-19:
So, the issue was with Sliver, not the Arduino, I went through so many host shields and Arduino's before figuring this out, the zeroes and Done! at the end mean that it actually worked.. The issue was that I was trying to use Sliver on my new MacBook which is running Catalina, once I ran Sliver on my older Macbook running High Sierra, it worked perfectly fine and I bypassed my 4s..

closed 2020-10-19 by Guinea1