a1exdandy/checkm8-a5 #4: Led stay still but device is not in PWN DFU mode

Kevdeme, 2020-07-29:
Hello, I get the 3 lights and then the lights stay still but the log says this:

https://pastebin.com/RRzZvFD5

Sliver fail with every option after that.

Any idea why am I not getting in to PWND DFU?

arslanmalik606, 2020-07-29:
I am also facing the same issue.. Led blibk 3 times and then after few second it comes solid. but my ipad mini and ipad 3 didnt in to pwnd dfu mode. plz help

amorozzo19, 2020-07-31:
Same problem here with an ipad mini 1st gen.

ramsdv31, 2020-08-06:
Exact same problem tried 4 USB host shields 1 arduino uno and 2 arduino megas and 4 cables Used xubuntu Linux And MacOs Mojave

mukeshbvn, 2020-08-11:
> Same problem here with an ipad mini 1st gen.

to all who have issue with not getting pwndfu must be on High Seirra.. i have same problem try on mojave,catlina but same error but after install High Sierra problem solve

arslanmalik606, 2020-08-12:
> > Same problem here with an ipad mini 1st gen.
> 
> to all who have issue with not getting pwndfu must be on High Seirra.. i have same problem try on mojave,catlina but same error but after install High Sierra problem solve

i have high serria in my mac. but still facing same problem. any solution ? if possible send me video of total process you have followed

ramsdv31, 2020-08-12:
> > Same problem here with an ipad mini 1st gen.
> 
> to all who have issue with not getting pwndfu must be on High Seirra.. i have same problem try on mojave,catlina but same error but after install High Sierra problem solve

Please post your arduino monitor. Do you get a bunch of ZEROS and a Done?

SargFowler, 2020-10-24:
Anyone got this to work?
I can't understand how the code compiles because the variable 'padding' is used at line 179 without being defined.

driversat, 2021-02-10:
.............
18:15:42.017 -> heap_feng_shui_req: setup status = 0, data status = 4
18:15:42.017 -> heap_feng_shui_req: setup status = 0, data status = 4
18:15:42.050 -> heap_feng_shui_req: setup status = 0, data status = 4
18:15:42.050 -> heap_feng_shui_req: setup status = 0, data status = 4
18:15:42.050 -> heap_feng_shui_req: setup status = 0, data status = 4
18:15:42.119 -> 2. set global state
then nothing