a1exdandy/checkm8-a5 #12: Not sure if this is the right output from the Arduino?

SargFowler, 2020-10-24:
I get the following output in the serial monitor from Arduino, but can't get the bypass tool to work as it says the device is not in pwned mode:

_checkm8 started
1. heap feng-shui
Stall status: 5
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 1, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 1, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
[... 629 lines cut ...]
    data: 0
    data: 0
    data: 0
    data: 0
3. heap occupation
heap_feng_shui_req: setup status = 0, data status = 0
overwrite sending ...
    SETUP: 0
    OUT (pre packet): 0
    OUT: 0
payload sending ...
    SETUP: 0
    OUT (pre packet): 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
Done!_

SargFowler, 2020-10-26:
This version of checkm8-A5 does not work for me, but the synackuk version does.
Perhaps there's some difference fixed in the synackuk version?
It's worth doing a diff between the two.

batelcho, 2020-11-19:
seems OK, but mine is not good .... any ideas 

19:10:39.085 -> heap_feng_shui_req: setup status = 0, data status = 4
19:10:39.085 -> heap_feng_shui_req: setup status = 0, data status = 4
19:10:39.085 -> heap_feng_shui_req: setup status = 0, data status = 4
.....................................................................
19:10:39.132 -> heap_feng_shui_req: setup status = 0, data status = 4
19:10:39.132 -> heap_feng_shui_req: setup status = 0, data status = 4
19:10:39.132 -> heap_feng_shui_req: setup status = 0, data status = 4
19:10:39.226 -> 2. set global state
19:10:39.226 -> OUT pre-packet: 0
19:10:39.226 -> Send random 0x40 bytes: 4
19:10:39.226 -> Send random 0x40 bytes HS: 5
19:10:39.226 -> Send zero length packet: 5
19:10:39.226 -> Send get status #1: 5
19:10:39.226 -> Send get status #2: 5
19:10:39.226 ->     data: 4
19:10:39.226 -> sending error
19:10:39.272 -> 3. heap occupation
19:10:39.272 -> heap_feng_shui_req: setup status = 0, data status = 0
19:10:39.318 -> overwrite sending ...
19:10:39.318 ->     SETUP: 0
19:10:39.318 ->     OUT (pre packet): 4
19:10:39.318 ->     OUT: 0
19:10:39.318 -> payload sending ...
19:10:39.318 ->     SETUP: 0
19:10:39.318 ->     OUT (pre packet): 0
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.318 ->     OUT: 4
19:10:39.365 -> Done!

SargFowler, 2020-11-19:
Have you tried the synackuk version?

batelcho, 2020-11-19:
Yes, but the difference with synackuk's repo is in the payload only and I'm having trouble at 'set global state' phase

FairXianyu, 2022-01-28:
Uhhh. . I have the same issue with my iPad3(3,1).I tried both a1ex's version and synackuk's version,but none of them work on my ipad.How did you end up solving the problem? If so, can you post the correct output, thanks

SargFowler, 2022-01-28:
What's your output?

FairXianyu, 2022-01-28:
> What's your output?

1. heap feng-shui
Stall status: 5
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heapfeng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shi_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup staus = 0, data stats = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 1, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup satus = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status= D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = D, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 1, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
heap_feng_shui_req: setup status = 0, data status = 4
[... 632 lines cut ...]
    data: 0
    data: 0
    data: 0
3. heap occupation
heap_feng_shui_req: setup status = 0, data status = 0
overwrite sending ...
    SETUP: 0
    OUT (pre packet): 0
    OUT: 0
payload sending ...
    SETUP: 0
    OUT (pre packet): 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
    OUT: 0
Done!

The same as yours

FairXianyu, 2022-01-28:
> What's your output?

Any clue?