a1exdandy/checkm8-a5 #14: ctrlReq_SETUP did you mean ctrlReq. Cant get passed this error to upload to Arduino.

vcopino, 2021-02-21:
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Uno"

In file included from /Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino:4:0:
sketch/constants.h:2:17: note: #pragma message: Building for A5 S5L8942X
 #pragma message "Building for A5 S5L8942X"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino: In function 'uint8_t heap_feng_shui_req(uint8_t, bool)':
checkm8-a5:113:21: error: 'class USB' has no member named 'ctrlReq_SETUP'; did you mean 'ctrlReq'?
   setup_rcode = Usb.ctrlReq_SETUP(0, 0, 0x80, 6, serial_idx, 3, 0x40a, sz);
                     ^~~~~~~~~~~~~
                     ctrlReq
/Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino: In function 'void set_global_state()':
checkm8-a5:159:15: error: 'class USB' has no member named 'ctrlReq_SETUP'; did you mean 'ctrlReq'?
   rcode = Usb.ctrlReq_SETUP(0, 0, 0x21, 1, 0, 0, 0, 0x40);
               ^~~~~~~~~~~~~
               ctrlReq
checkm8-a5:178:61: error: no matching function for call to 'USB::ctrlReq(int, int, int, int, int, int, int, int)'
   rcode = Usb.ctrlReq(0, 0, 0x21, 1, 0, 0, 0, padding + 0x40);
                                                             ^
In file included from /Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:44:0,
                 from /Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino:1:
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note: candidate: uint8_t USB::ctrlReq(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t*, USBReadParser*)
         uint8_t ctrlReq(uint8_t addr, uint8_t ep, uint8_t bmReqType, uint8_t bRequest, uint8_t wValLo, uint8_t wValHi,
                 ^~~~~~~
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note:   candidate expects 11 arguments, 8 provided
/Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino: In function 'void heap_occupation()':
checkm8-a5:206:48: error: no matching function for call to 'USB::ctrlReq(int, int, int, int, int, int, int, int)'
   rcode = Usb.ctrlReq(0, 0, 0, 0, 0, 0, 0, 0x40);
                                                ^
In file included from /Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:44:0,
                 from /Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino:1:
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note: candidate: uint8_t USB::ctrlReq(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t*, USBReadParser*)
         uint8_t ctrlReq(uint8_t addr, uint8_t ep, uint8_t bmReqType, uint8_t bRequest, uint8_t wValLo, uint8_t wValHi,
                 ^~~~~~~
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note:   candidate expects 11 arguments, 8 provided
checkm8-a5:218:62: error: no matching function for call to 'USB::ctrlReq(int, int, int, int, int, int, int, unsigned int)'
   rcode = Usb.ctrlReq(0, 0, 0x21, 1, 0, 0, 0, sizeof(payload));
                                                              ^
In file included from /Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:44:0,
                 from /Users/victorcopino/Downloads/checkm8-a5-master/checkm8-a5/checkm8-a5.ino:1:
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note: candidate: uint8_t USB::ctrlReq(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t*, USBReadParser*)
         uint8_t ctrlReq(uint8_t addr, uint8_t ep, uint8_t bmReqType, uint8_t bRequest, uint8_t wValLo, uint8_t wValHi,
                 ^~~~~~~
/Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/UsbCore.h:271:17: note:   candidate expects 11 arguments, 8 provided
Multiple libraries were found for "Usb.h"
 Used: /Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0
 Not used: /Users/victorcopino/Documents/Arduino/libraries/USB_Host_Shield_2.0
exit status 1
'class USB' has no member named 'ctrlReq_SETUP'; did you mean 'ctrlReq'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Jeanchristian47, 2021-03-17:
Hello,do you resolve this ?

gintechsystems, 2021-12-22:
Patching usb_shield should resolve this error, as shown on the github homepage for this project:

**USB Host Library Rev. 2.0 installation and patching**

```
cd path/to/Arduino/libraries
git clone https://github.com/felis/USB_Host_Shield_2.0
cd USB_Host_Shield_2.0
git checkout cd87628af4a693eeafe1bf04486cf86ba01d29b8
git apply path/to/usb_host_library.patch
```

meetanthony, 2023-08-01:
don't do update usb_host_shield library in arduino ide. and that's all )