a1exdandy/checkm8-a5 #25: I can't enter PWNED DFU on 4s with Arduino UNO and USB Host Shield.

mint0mango, 2023-07-31:
Greetings, a few weeks or months ago, I started researching icloud bypass methods for A5 devices. After trying all methods (which I read are compatible) I bought an arduino uno with Atmega16U2 usb chip and a host shield with MAX3421E chip. I got an error in Arduino IDE when I tried with Synackuk or A1exdandy checkm8. Then I learned that I had to solder 3 jumpers on the Host Shield and change the CPID number. In my attempts after soldering and changing the number, the .ino file was successfully written, but when I connected the iPhone 4S model device to the Host Shield, the LED did not flash (by the way, I connected the negative part of the LED to the first 2nd port and the positive part to the other 2nd port.). I also removed the LED and touched it to a battery and it was working. When I plugged it into the USB host shield with the iPhone turned on normally, I saw that it was not charging, but when I connected the LED to a USB cable and plugged it in, I saw that it was on. Just to make sure one more time, I plugged my Android phone into the host shield and it charged too. I then tried with a new 24 pin Lightning Cable but it didn't charge either. Ignoring the problems such as the LED not lit and the iPhone not charging, I started the process to enter the PWNED DFU, but none of the Alternate Ramdisk, Standard Ramdisk and ibss only methods were successful. (In other words, the screen of the device remained in DFU mode and did not switch to the verbose screen.) I used a hackintosh system installed in Monterey to perform these operations. If there is any information or system feature you want, feel free to ask. Google Translate was used to write this text, sorry for any errors. (I took this video as a guide https://www.youtube.com/watch?v=v_fWwCaGWU8)

lizbeth-mutterhunt, 2023-11-28:
I have a A5 iPhone4s, with a A5_8940, Legacy-iOS-Kit writes with downgrade:

Extracting iBSS.n94.RELEASE.dfu...
Sending iBSS (78044 bytes)...
ERROR: Unable to send iBSS component: Unable to upload data to device
ERROR: Unable to send iBSS to device
ERROR: Unable to place device into recovery mode from DFU mode

missing key for this; hackintoshing used to be my hobby, but now I do mac hardware and open it up for all systems.

nrnsbr, 2024-06-29:
exactly the same here. tried everything (ipod 5th and ipad3, 8945 & 8942) but the led won't turn on, devices won't enter in pwned dfu.

mutterhunt78, 2024-06-30:
in the meantime the accumulator gave up while entering DFU mode, I guess it is a special compilation and protection on the alpine model of the 4s running around Germany, France, Swiss, Austria, etc.... the Arduino is now plain stuff here, nobody uses it and I dont know if I really need it again, although it is a good concept for a microprocessor.